FYSETC/FYSETC-Mini-12864-Panel #1: Display does not work

Brainc0re, 2019-04-04:
I have a CR10S-Pro which normally use a touchscreen. But i didn't like and replaced it with a 12864 reprap full graphic controller. That controller/display works since weeks. So i thought, that this mini12864 will work too. But im not able to run it :/ Not even the backlight works (Pin1 and Pin2 are correct - i can meassure 5V - Pin1 (+) // Pin2 (-) and J2 has 3.3V)
I made the changes in Marlin which are provided by fysetc on their website.

Is it possible, that this display wont work on the creality mainboard?

Current Firmware (reprap 12864): https://drive.google.com/open?id=1HoQVlum-DM5lGhCuu1yqCfW6UupuqbK6
FW for mini12864: https://drive.google.com/open?id=11KdF0nT0fa-EBrvGuiYhUTILRo4qJgjR
Mainboard layout: https://drive.google.com/open?id=1Ya5zHF_WQpw7XY_BoL0zs20UwexBvtwI

Attention: Both FWs has the same *.ino name but different folder names.

ejhoness, 2021-12-17:
180º EXP1 and EXP2 in lcd,

Eugeniusz-Gienek, 2024-03-12:
> 180º EXP1 and EXP2 in lcd,

OMG that worked! Thanks man!

closed 2024-09-19 by marcoa131