FYSETC/FYSETC-Mini-12864-Panel #2: STEP files for display are corrupted

jdlongenecker, 2019-07-28:
I am making an enclosure for this display, but the STEP files included in the Github links will not open. Can you check the files and reupload them?

Thanks!

jonathanlundstrom, 2019-09-08:
I've been at this for months and still haven't been able to use these files. The LCD mount is one of the last things on my to-do in order to finish my machine and I have not been able to design it properly. Please upload new files!

danym21, 2019-09-17:
looks like some one has changed the step files to a .zip file with a link to a 
Potentially Unwanted Application (PUA)
https://www.f-secure.com/sw-desc/pua.shtml
please replace it with the correct files!!

jonathanlundstrom, 2019-09-17:
I got the correct file from a guy on their Facebook group and used it to design an enclosure for my v2.1 LCD. It worked really well and the dimensions were very accurate although there are some differences in the model of the potentiometer and the reset button, but it's all good.

I uploded it to my website so you can download it as well: [Download here](http://jonathanlundstrom.me/download/325/) :)

glabifrons, 2019-10-29:
FYSETC: The ZIP files you uploaded are not usable without installing malware. Please delete the zip files you created with KuaiZip and replace them with files created with a conventional non-proprietary ZIP program (such as InfoZip's open-source ZIP software).

Or simply upload the STEP file uncompressed, it's the same size as the zip you uploaded.

bipsendk, 2020-06-28:
Used a VM to install the PUA Zip application and extracted the STEP files for v2.0 and 2.1 - they are on https://drive.google.com/drive/folders/1SBvUNJBHfiaKfwsOxPWEyJ8HOudN6aJK?usp=sharing

wiredrat, 2021-02-21:
I don't think that forcing your customers to download a trojan is the rigth company policy.

https://www.virustotal.com/gui/file/2f187bdf55d5ff249aced33c66b4f7149e9a6972aed7bd61d15e98438ae51824/detection

3DFabrik, 2021-04-08:
> I got the correct file from a guy on their Facebook group and used it to design an enclosure for my v2.1 LCD. It worked really well and the dimensions were very accurate although there are some [...]

Thanks Mate!

marcoa131, 2024-09-19:
I've deleted the .zip files for 2.0 and 2.1 and added the corresponding .step files.

closed 2024-09-19 by marcoa131

marcoa131, 2024-09-20:
> > I've deleted the .zip files for 2.0 and 2.1 and added the corresponding .step files.
> 
> ??? The last commit was a year ago, the zip files are still there and there are no new files.

I did the commit through github desktop, it takes a while until it updates on the web.
[image]

glabifrons, 2024-09-20:
> I did the commit through github desktop, it takes a while until it updates on the web. ![image]

Wow! You're not kidding that it's slow... it just showed up (4 hours later).

Thanks!