FYSETC/FYSETC-Mini-12864-Panel #3: Configuration update

Druckbar, 2020-11-10:
The configuration steps are outdated and misleading. Marlin 2.0.x implements the FYSETC variants natively.

There is no need to edit the `pins_RAMPS.h`.

In `configuration.h` search for **FYSETC_MINI_12864** and enable your revision of the display.

Follow the steps for your board revision and either enable **RGB_LED** (< 2.1) or **NEOPIXEL_LED** (>= 2.1).

I recommend enabling the **LED_CONTROL_MENU** in `configuration_adv.h` since you can control brightness from the LCD menu and you can set your preferred colors on startup.

Any further modifications are not neccessary. The brightness should be automatically set to 220.

I've tested the procedure with current Marlin release 2.0.7.2 on a BTT SKR 1.4 Turbo and a MKS Gen L 1.0.

YoukiNya, 2021-04-02:
I ues Marlin 2.0.7.2 on MKS SgenL 1.0 
The screen can display bootscreen , but motherboard reboot again and again. 
I don't know what was wrong

romass, 2021-11-11:
I have a similar problem.
LCD lights up, but no visible text appears.
Any help would be appreciated. 

Board: MKS Gen L V2.1
Firmware: Marlin 2.0.9x
LCD: Fysetc Mini 12864 V2.1

configs: https://github.com/romass/3D-Printing/tree/main/MKS_gen_L_V21_Fysetc%20Mini%2012864%20V2.1%20Marlin%202

pmead7399, 2022-04-24:
If anyone is left here I need help too. I have a 2.1 and have it enabled in Marlin 2.0.1 and I get to the FINAL step which is to find the ultralcd_something_something and the file doesn't exist. Set to 220 by default? Should I be using exp 1 or 2 or both? Please help.

Obs1234, 2024-02-03:
Druckbar...Found this post. Was working on getting my BTT 12864 V2 running with no luck. Your simple how-to made my day!! Cheers from Canada, EH?

Obs1234, 2024-02-05:
Also adding this to clarify all above setting!!!

https://www.reddit.com/r/BIGTREETECH/comments/10i2ewm/fysetc_mini_12864_v21_rgb_octopus_v11_marlin_21x/