FYSETC/FYSETC-Mini-12864-Panel #6: TH3D-EZBOARD LITE

spankurmonkey, 2021-01-16:
Do you know how to make the Fysetc Mini 12864 to work with the TH3D-EZBOARD LITE using their unified firmware?

spankurmonkey, 2021-01-17:
You only marked 8 of the 10 pins on the TH3D EzBoard Lite set up.   I think you missed the V+ and GND

spankurmonkey, 2021-01-17:
Also TH3D has a expansion header - not sure if that would help in this situation or not.

https://www.th3dstudio.com/product/ezboard-creality-dual-lcd-adapter-board/

Also, instead of giving a bin file for the firmware it would be best to advise what to change in the firmware to make the LCD work.  That way if there is new firmware then you would not need to keep updating it.   Or you could do both, for those who want to update it just show what files, changes to make and other who do not then list the firmware and what version of it is.

[image]
[image]
[image]

spankurmonkey, 2021-01-19:
I tried to compile the unified firmware with your instructions in the pins file and configuration.   I get the following errors

`Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_filament.cpp.o
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_game.cpp.o
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_info.cpp.o
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_main.cpp.o
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_media.cpp.o
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:42:
Marlin\src\lcd\dogm\ultralcd_DOGM.h:60:36: error: 'LCD_PINS_ENABLE' was not declared in this scope
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                    ^~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:60:53: error: 'LCD_PINS_RS' was not declared in this scope; did you mean 'LCD_PINS_D4'?
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                                     ^~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_mixer.cpp.o
*** [.pioenvs\LPC1769\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1
=================================================================================== [FAILED] Took 41.48 seconds ===================================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:41.477
============================================================================== 1 failed, 0 succeeded in 00:00:41.477 ==============================================================================The terminal process "C:\Users\kswan\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

spankurmonkey, 2021-01-19:
Here is how I have the pins file attached and then in the configuration.h I have this

`//
// FYSETC variant of the MINI12864 graphic controller with SD support
// https://wiki.fysetc.com/Mini12864_Panel/
//
#define FYSETC_MINI_12864_X_X    // Type C/D/E/F. No tunable RGB Backlight by default
//#define FYSETC_MINI_12864_1_2    // Type C/D/E/F. Simple RGB Backlight (always on)
//#define FYSETC_MINI_12864_2_0    // Type A/B. Discreet RGB Backlight
//#define FYSETC_MINI_12864_2_1    // Type A/B. NeoPixel RGB Backlight
//#define FYSETC_GENERIC_12864_1_1 // Larger display with basic ON/OFF backlight.`

[pins_TH3D_EZBOARD.txt](https://github.com/FYSETC/FYSETC-Mini-12864-Panel/files/5839086/pins_TH3D_EZBOARD.txt)

spankurmonkey, 2021-01-19:
images from VScode
[image]
[image]

TuguanLing, 2021-01-20:
You can try to add
      #define LCD_PINS_ENABLE                 P0_18
      #define LCD_PINS_RS                     P0_16

TuguanLing, 2021-01-20:
[image]

spankurmonkey, 2021-01-20:
it errored again 

`> Executing task: C:\Users\kswan\.platformio\penv\Scripts\pio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.5
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\watchdog.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\servo.cpp.o
Compiling .pioenvs\LPC1769\src\src\MarlinCore.cpp.o
Compiling .pioenvs\LPC1769\src\src\core\serial.cpp.o
Compiling .pioenvs\LPC1769\src\src\core\utility.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\controllerfan.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\pause.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\tmc_util.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\abl\G29.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\abl\M421.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\calibrate\G28.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\calibrate\M48.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M200-M205.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M220.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M221.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M301.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M302.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M92.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M111.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M120_M121.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M17_M18_M84.cpp.o
[... 83 lines cut ...]
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_filament.cpp.o
In file included from Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:42:
Marlin\src\lcd\dogm\ultralcd_DOGM.h:60:36: error: 'LCD_PINS_ENABLE' was not declared in this scope
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                    ^~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.h:60:53: error: 'LCD_PINS_RS' was not declared in this scope; did you mean 'LCD_PINS_D4'?
   60 |     #define U8G_PARAM LCD_PINS_D4, LCD_PINS_ENABLE, LCD_PINS_RS // AVR version ignores these pin settings
      |                                                     ^~~~~~~~~~~
Marlin\src\lcd\dogm\ultralcd_DOGM.cpp:78:15: note: in expansion of macro 'U8G_PARAM'
   78 | U8G_CLASS u8g(U8G_PARAM);
      |               ^~~~~~~~~
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_game.cpp.o
Compiling .pioenvs\LPC1769\src\src\lcd\menu\menu_info.cpp.o
*** [.pioenvs\LPC1769\src\src\lcd\dogm\ultralcd_DOGM.cpp.o] Error 1
[... cut ...] ====
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:19.336
======================================================================================== 1 failed, 0 succeeded in 00:00:19.336 ========================================================================================The terminal process "C:\Users\kswan\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

spankurmonkey, 2021-01-20:
Is the issue this:

in configuration.h  it says '  **#define FYSETC_MINI_12864_X_X**   '

however, in the pins_TH3D_EZBOARD.h it says '  **#elif ENABLED(FYSETC_MINI_12864)**  '

Should it really be ' **#elif ENABLED(FYSETC_MINI_12864_X_X)** '  ??

spankurmonkey, 2021-01-20:
I tried switching it didn't make a difference.

I did notice that this new section we added into the pins_TH3D_EZBOARD.h is active in your screen print but in mine its dimmed out....  see photo here.

dimmed out after the **#if HAS_SPI_LCD**

[image]

spankurmonkey, 2021-01-20:
I added those two lines - didn't make a difference.

Then I looked and the if statements and I wonder if that is the issue... should the HAS_SPI_LCD be at the end and not the beginning?

I used this code below and it compiled 

` #if ENABLED(CR10_STOCKDISPLAY)

    #define BEEPER_PIN                       P1_31
    #define BTN_EN1                          P3_26
    #define BTN_EN2                          P3_25
    #define BTN_ENC                          P1_30
    #define LCD_PINS_RS                      P0_16
    #define LCD_PINS_ENABLE                  P0_18
    #define LCD_PINS_D4                      P0_15
    #define KILL_PIN                         P2_11

  #elif ENABLED(FYSETC_MINI_12864)

    #define BTN_EN1                         P3_26
    #define BTN_EN2                         P3_25
    #define BTN_ENC                         P1_30

    #define DOGLCD_CS                       P0_18
    #define DOGLCD_A0                       P0_16
    #define DOGLCD_SCK                      P1_31
    #define DOGLCD_MOSI                     P0_15
    #define LCD_PINS_ENABLE                 P0_18
    #define LCD_PINS_RS                     P0_16

    // #define KILL_PIN                        -1
    // #define BEEPER_PIN                      -1 
    // #define LCD_PINS_ENABLE                 -1
    // #define LCD_PINS_D4                     -1

    // #define LCD_BACKLIGHT_PIN               -1

    #define FORCE_SOFT_SPI

    #define LCD_RESET_PIN                   P2_11

    // #if EITHER(FYSETC_MINI_12864_1_2, FYSETC_MINI_12864_2_0)
    //   #ifndef RGB_LED_R_PIN
    //     #define RGB_LED_R_PIN               P0_18
    //   #endif
    //   #ifndef RGB_LED_G_PIN
    //     #define RGB_LED_G_PIN               P0_18
    //   #endif
    //   #ifndef RGB_LED_B_PIN
    //     #define RGB_LED_B_PIN               P0_18
    //   #endif
    // #elif ENABLED(FYSETC_MINI_12864_2_1)
    //   #define NEOPIXEL_PIN                  P0_18
    // #endif
    
  **#elif HAS_SPI_LCD**
    #error "Only the CR10_STOCKDISPLAY is supported with TH3D EZBoard."
  #endif`

spankurmonkey, 2021-01-20:
`> Executing task: C:\Users\kswan\.platformio\penv\Scripts\pio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.5
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\tft\xpt2046.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\timers.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\LPC1768\watchdog.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\HAL_spi_L6470.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\eeprom_if_spi.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pioenvs\LPC1769\src\src\HAL\shared\servo.cpp.o
Compiling .pioenvs\LPC1769\src\src\MarlinCore.cpp.o
Compiling .pioenvs\LPC1769\src\src\core\serial.cpp.o
Compiling .pioenvs\LPC1769\src\src\core\utility.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\controllerfan.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\pause.cpp.o
Compiling .pioenvs\LPC1769\src\src\feature\tmc_util.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\abl\G29.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\bedlevel\abl\M421.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\calibrate\G28.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\calibrate\M48.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M200-M205.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M220.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M221.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M301.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M302.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\config\M92.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M108_M112_M410.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M111.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M120_M121.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M17_M18_M84.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M211.cpp.o
Compiling .pioenvs\LPC1769\src\src\gcode\control\M280.cpp.o
[... 370 lines cut ...]
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.1
|-- <U8glib-HAL> 0.4.3
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Checking size .pioenvs\LPC1769\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  51.8% (used 16948 bytes from 32736 bytes)
Flash: [====      ]  44.6% (used 211848 bytes from 475136 bytes)
============================================================================================= [SUCCESS] Took 3.77 seconds =============================================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        SUCCESS   00:00:03.767
============================================================================================= 1 succeeded in 00:00:03.767 =============================================================================================
Terminal will be reused by tasks, press any key to close it.
`

spankurmonkey, 2021-01-20:
Looks like there is one warning

Compiling .pioenvs\LPC1769\lib533\U8glib-HAL\clib\u8g_com_psoc5_ssd_hw_parallel.c.o
.piolibdeps\LPC1769\U8glib-HAL\src\clib\u8g_com_i2c.c:44:16: warning: 'u8g_i2c_opt' defined but not used [-Wunused-variable]
44 | static uint8_t u8g_i2c_opt; /* U8G_I2C_OPT_NO_ACK, SAM: U8G_I2C_OPT_DEV_1 */
| ^~~~~~~~~~~

TuguanLing, 2021-01-20:
Whether the Marlin firmware you use is the latest, you can check whether the configuration in ‘Configuration.h’ is correct and whether other irrelevant macro definitions are turned on; if this configuration can be compiled and passed, you can try to drive mini12864.

spankurmonkey, 2021-01-20:
Hi there -  did you look at your posted changes and did you determine if the **#elif HAS_SPI_LCD** has the correct placement in the code... 

Only when I changed it did it compile.      If that is correct, it has the wrong placement can you correct the example page so that if others come to do the same they have the correct code for firmware changes to drive the Fysetc Mini 12864 ?

TuguanLing, 2021-01-21:
Thank you for your suggestion, I have revised the changes posted

But I can pass the test before and after the modification, I am now trying to find the problem