FYSETC/FYSETC-Mini-12864-Panel #10: Screen does not activate/update until click button

danbrianwhite, 2021-07-19:
When using Marlin, mks robin e3 board and a Fysetc mini display, the display does not automatically.change from the "Marlin" version screen. It gets stuck and needs a rotary button press. Then once button pressed, it goes to info screen. 

Sometimes, the screen does not update after button reset pressed. And is "stuck" on previous printing screen until you press.the rotary button.

Any ideas of what is wrong or what code needs to be added to Marlin to fix this?

I can code it if I get a hint or direction front he Fysetc devs.

Best,
Dan

kjanek, 2021-07-23:
Could you send me a link to a guide how you have made this screen almost working on the mks robin e3. Because I have a MKS Robin E3D and cant figure out how to set things correctly in marlin, so that it shows text on screen (neopixel is working - i have a 2.1 version of the screen) but the screen stays blank.

mks-viva, 2021-08-10:
@danbrianwhite @kjanek 
https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/FYSETC_BTT_MINI12864

ejhoness, 2021-12-17:
"Connect the EXP1 and EXP2 cable to the screen and to the motherboard if it still doesn't work this means you have to rotate the plug 180 degrees."

closed 2026-02-04 by Rushmere3D